leaphy-robotics/leaphy-webbased #38: Exportbutton - Serial monitor tot csv-file

Roelandsmith, 2023-09-20:
Great tool for Science-education: Export log-file Serial monitor to .csv-file (excel-compatible)
Button could be placed alongside the bin-button.

JeftadeGraaf, 2023-12-10:
Ik heb inmiddels de support toegevoegd, ik heb de knop maar toegevoegd bij de serial window. Leek me een betere plek.

issues?
- De file name is nu een altijd dezelfde naam
- Je kan nu nog downloaden ook al is er nog geen serial output
- Nog andere fouten gevonden?

JeftadeGraaf, 2023-12-15:
[image]
Bedoel je daarbij de waardes tussen de tijd en data in? Ik zie trouwens niet je voorstel voor een filename.

rmoesbergen, 2023-12-15:
> ![image](https://private-user-images.githubusercontent.com/91546552/290917772-fe5b0f7e-01b4-44b1-b8c5-b34c0a6a532d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

Nee, ik bedoel in de csv zit in de data kolom na elke regel een extra \n, want die wordt door de arduino meegestuurd.

De comments staan hier:
https://github.com/leaphy-robotics/leaphy-webbased/pull/98/files

JeftadeGraaf, 2023-12-27:
Heb de \n gefixt. Zie nog steeds niet de comments. https://github.com/leaphy-robotics/leaphy-webbased/pull/98/files hier staan alleen de veranderde files, zonder comments

rmoesbergen, 2023-12-28:
Hmm, apart, komt misschien omdat ik de PR geopend heb, en er zelf comments in heb gezet. Ik gooi ze wel hier:

[image]

[image]

rmoesbergen, 2023-12-28:
Misschien beter als jij een PR opent voor die branch. Zowieso wel handig, anders staat ie op mijn naam.

JeftadeGraaf, 2023-12-28:
yes, gefixt. nu heb PR aangemaakt

closed 2023-12-28 by JeftadeGraaf

rmoesbergen, 2023-12-31:
👍  wat mij betreft kan deze gemerged!