leaphy-robotics/leaphy-webbased #53: Categorieen sensor en actuator in Original en Uno omgeving (alle omgevingen behalve Flitz)

JeftadeGraaf, 2023-10-10:


rmoesbergen, 2023-10-12:
Ik weet niet wat er precies is aangepast, maar ik zie nog niet het gewenste effect. De omschrijving is ook wat wazig. Volgens mij was het de bedoeling dat voor alle robots behalve de Flitz, we de toolbar hetzelfde inrichten zoals de Nano: dus met een Sensoren en Actuatoren item in de toolbar. Als ik nu naar de Uno ga, zie ik nog steeds 'Arduino Uno' en 'Leaphy Extra' als categorieën staan.

De tweede aanpassing was om Sensoren en Actuatoren onder elkaar te plaatsen, als eerste 2 categorieën. Nu staat de ene helemaal bovenaan, en de ander helemaal onderin de toolbox.

Hopelijk maakt dit het wat duidelijker. @JeftadeGraaf zou je hier nog eens naar willen kijken>

koen1711, 2023-10-28:
@JeftadeGraaf 

Wil je dat ik er naar kijk?

rmoesbergen, 2023-11-17:
@JeftadeGraaf Ik heb getest en zie:

- Bij alle robots ziet de toolbar er netjes uit, sensors/actuatoren staat mooi bovenaan, nice. Alleen bij alle robots behalve de Nano zijn de blocks nog niet goed gesorteerd in de juiste categorie. Bij de Uno bijvoorbeeld staan er nog sensoren in actuatoren categorie en andersom.
- De Flitz (zowel Uno als Nano) zou hetzelfde moeten blijven, alleen is nu de bovenste categorie ook 'Sensors' geworden. Voor beide Flitz-en zou dat niet aangepast moeten worden (dus gewoon 'Leaphy Flitz', of wat er ook stond).

Kun je deze 2 dingen nog aanpassen?

JeftadeGraaf, 2023-11-20:
Heb het nu aangepast, zou je er nog eens naar willen kijken?

rmoesbergen, 2023-11-20:
> Heb het nu aangepast, zou je er nog eens naar willen kijken?

ja, nu is het perfect! 🎉

closed 2023-11-21 by JeftadeGraaf