leaphy-robotics/leaphy-webbased #54: Pinnen van commandoblokken handsensor (out op A0), RGB (gnd D8, R D9, G D10, B D11) en buiksensor (out op D2) aanpassen

JeftadeGraaf, 2023-10-10:


rmoesbergen, 2023-10-12:
Belangrijk dat dit alleen voor de nieuwe Flitz nano gebeurt, de Uno moet blijven zoals het is.

rmoesbergen, 2023-10-20:
@beekumo De huidige handsensor gebruikt deze pins:

```
        if(sensorType == "1"){
            setup = 'pinMode(8, OUTPUT);\n  pinMode(9, OUTPUT);\n  pinMode(10, INPUT);\n  digitalWrite(8, LOW);\n  digitalWrite(9, HIGH);\n'
            code = 'digitalRead(10)'
        } else if(sensorType == "2"){
            setup = 'pinMode(8, INPUT);\n  pinMode(9, OUTPUT);\n  pinMode(10, OUTPUT);\n  digitalWrite(8, LOW);\n  digitalWrite(9, HIGH);\n'
            code = 'digitalRead(8)'
        }
```
Er zijn dus 2 'out' pins. Welke van de 2 moet naar A0?
Daarnaast zijn er nog 2 types handsensor, deze ene leest pin 10 en de ander pin 8. In het block zit een drop-down keuze voor welk type het is. Moet dit bij de nano-versie ook zo blijven?

rmoesbergen, 2023-10-20:
@beekumo zelfde vraag over de handsensor:
```
    Arduino.forBlock['leaphy_flitz_read_hand_sensor'] = function (block){
        Arduino.setups_['setup_flitz_hand'] = 'pinMode(14, OUTPUT);\n pinMode(15, OUTPUT);\n pinMode(2, INPUT);\n digitalWrite(14, HIGH);\n digitalWrite(15, LOW);\n'
        var code = 'analogRead(2)'
        return [code, Arduino.ORDER_ATOMIC];
    };
```
welke van de output pins moet hier naar A0?

rmoesbergen, 2023-10-21:
aangepast in #71

rmoesbergen, 2023-10-21:
released in 1.6.0

closed 2023-10-21 by rmoesbergen