leaphy-robotics/leaphy-webbased #64: Automatische browser based tests maken

rmoesbergen, 2023-10-10:
Testen neem veel tijd in beslag, en hoe meer we toevoegen hoe langer het duurt en hoe meer fouten we gaan missen. Misschien goed om tijd te investeren in een brower based test framework zoals Selenium, Cypress, Playwright, etc.. Klein beginnen.

rmoesbergen, 2024-02-29:
Stap 1: Uitvinden van diverse tools + pro's/cons
Stap 2: Implementeren van de tool in een github action (basis opzet met een paar simpele tests).

pajotg, 2024-03-20:
Ik heb ondertussen Selenium en Playwright uitgeprobeert en 1 test gemaakt van een simpele bug die ik tegenkwam om te zien hoe alles werkt.
Ik wil nog Cypress en Protractor verder bekijken en daarna kunnen we een framework kiezen die we gaan gebruiken.

Selenium:
* pros:
  * kan erg makkelijk tests maken met de browser IDE
  * blijkbaar erg populair dus veel over te vinden online
* cons:
  * Niet compleet overzichtelijk wat er exact gebeurt, veel verstopt
  * Vreemd genoeg erg laggy, geen idee waarom
  * De browser IDE maakt niet telkens een nieuwe user profile aan en dus dingen zoals de "choose language" / "changelog" zijn er niet in locale tests, maar wel in de ge-automatiseerde runs

Playwright:
* pros:
  * gebruikt yarn dus hoef niet nieuwe dingen te installeren
  * de VSCode extension heeft goeie tools om tests te maken (recorden van acties + geeft highlighting van wat de regel doet die je zit te editen)
  * Gebruikt TS voor de tests dus als je TS kent is het erg makkelijk te leren
* cons:
  * Gebruikt TS voor de tests dus als je geen TS kent zou het moeilijker zijn om tests te maken, maar is nogsteeds te doen met de recording tool
  * geen support voor oude browsers
  * mobile devices zijn minder goed ondersteund

pajotg, 2024-03-27:
Cypress:
* pros:
  * De tests maken snapshots dus je kan terug kijken wat er gebeurt was in in elke test
  * Kan tests opslaan en draaien in "de cloud", kan dus test resultaten delen met het hele team
* cons
  * De tool waarmee je tests record is nog niet compleet en onder "experimental", is wat moeilijker om tests mee te schrijven
  * Geen directe feedback op de test die je zit te schijven, je moet de hele test opnieuw draaien om de resultaten te zien
  * De overlays in de replay/snapshots zijn een beetje buggy, dus is soms wat moeilijk om te zien wat er echt gebeurt

Protractor: is end-of-life, heb niet verder gekeken.

Ik denk dat Playwright de beste keuze is.
Het is vergelijkbaar met Cypress kwa functionaliteit, maar is veel makkelijker om tests mee te maken en tooling is gewoon beter.
Cypress kan "de cloud" gebruiken om tests te draaien, maar misschien is dat niet nodig hebben als we Github actions gebruiken.

koen1711, 2024-03-28:
Ik ben het eens met Playwright ook omdat de limitations niet echt limiting zijn, oudere browsers supporten ook geen web serial en mobile supporten we helemaal niet op het moment.

closed 2024-04-14 by koen1711