leaphy-robotics/leaphy-webbased #69: Pins aanpassen voor motorshield commandoblock met nano

rmoesbergen, 2023-10-16:


rmoesbergen, 2023-10-16:
[image]

rmoesbergen, 2023-10-16:
De pin layout is anders dan voor het L293P shield. Zie screenshot voor de goede layout. Let op: dit is dus alleen voor de nano motorshield, het origineel moet blijven zoals het nu is.

rmoesbergen, 2023-10-20:
@beekumo De huidige shield heeft de volgende pin layout:
```
// PIN definitions for Fundumoto L298P motor shield with built in buzzer
#define M1_PWM          (10)    // Motor A PWM
#define M1_DIR          (12)    // Motor A direction
#define M2_PWM          (11)    // Motor B PWM
#define M2_DIR          (13)    // Motor B direction
#define TONE_PIN        (4)     // DIG-OUT
#define LED1_RED        (6)     // PWM-OUT
#define LED1_GREEN      (5)     // PWM-OUT
#define LED1_BLUE       (3)     // PWM-OUT
#define LINE_LEFT       (16)    // DIG-IN       A2 tinkerkit In2
#define LINE_RIGHT      (17)    // DIG-IN       A3 tinkerkit In3
#define US_ECHO         (8)     // DIG-IN/OUT
#define US_TRIG         (7)     // DIG-OUT
```

Hoe ziet de nieuwe layout eruit? Uit de screenshot kan ik niet goed opmaken welke pins er nu anders zijn.

beekumo, 2023-10-20:
@rmoesbergen hierbij de layout voor het nieuwe motor shield, alleen voor in Flitz nano en Leaphy original nano omgeving(!)

#define M1_PWM          (3)    // Motor A PWM
#define M1_DIR          (2)    // Motor A direction
#define M2_PWM          (11)    // Motor B PWM
#define M2_DIR          (4)    // Motor B direction

rmoesbergen, 2023-10-28:
Aangepast en gereleased in 1.7.0.

closed 2023-10-28 by rmoesbergen