leaphy-robotics/leaphy-webbased #104: Easybloqs suitable for original Arduino computers

beekumo, 2024-01-03:
Webbased Leaphy easybloqs can not be used for original Arduino computers like nano every, would be nice to have this feature available

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

koen1711, 2024-03-08:
#166

koen1711, 2024-03-23:
The uploader itself has already been made, just needs an frontend for it.

koen1711, 2024-04-15:
I think this can be marked as completed as we added a lot more robots, if we are missing any let us know and we will add those.

closed 2024-04-15 by koen1711