leaphy-robotics/leaphy-webbased #114: Uploaden geeft locking error na 1e upload

rmoesbergen, 2024-01-26:


rmoesbergen, 2024-01-26:
@koen1711 dit gaat mis sinds de python editor, zie ook de melding over de chromebooks in whatsapp. Enig idee?

rmoesbergen, 2024-01-26:
```
            upload(t, e=(i=>{}
            )) {
                var i = this;
                return Qt(function*() {
                    if (i.isUploading)
                        throw new Error("Arduino is already uploading");
                    let r;
                    i.robotWiredState.clearUploadLog(),
                    i.isUploading = !0,
                    i.readStream = i.port.readable.getReader(),
                    i.writeStream = i.port.writable.getWriter();
                    try {
```

Die i.port.readable.getReader() gooit deze error. Blijkbaar wordt de poort niet unlocked ergens.

koen1711, 2024-01-26:
Hmmm, ik heb het vaker al zien langs komen maar het gebeurt soms wel soms niet zal er naar kijken.

rmoesbergen, 2024-01-26:
> Hmmm, ik heb het vaker al zien langs komen maar het gebeurt soms wel soms niet zal er naar kijken.

oke, thanks. Ik kan het hier iig consistent reproduceren.

koen1711, 2024-01-26:
fixed

closed 2024-01-26 by koen1711