leaphy-robotics/leaphy-webbased #133: Blocks redefining.

JeftadeGraaf, 2024-02-01:
When we are working in a workspace and decide to start a new one. All the blocks are redefined, causing a lot of errors in the console.log.

[image]

koen1711, 2024-03-08:
Not really a bug just not pretty

koen1711, 2024-04-15:
Closing this as it's not necessary and a normal user would not open the console.

koen1711, 2024-05-01:
Fixed as of #246

rmoesbergen, 2024-05-01:
Pls stop closing other peoples issues

koen1711, 2024-05-02:
Nvm, I thought it was fixed and it could be fixed within seconds.
[image]
It can be fixed by just removing one of these.