leaphy-robotics/leaphy-webbased #165: Upload werkt niet op linux

Vosjedev, 2024-03-02:
**Describe the bug**
Tijdens het uploaden op linux komt de volgende error in de js console, waarna de popup zich aanpast:
```
TypeError: Cannot read properties of null (reading 'releaseLock')
    at tN.<anonymous> (main.RMJQDME5.js:1954:8020)
    at Generator.next (<anonymous>)
    at main.RMJQDME5.js:1:1395
    at new q (main.RMJQDME5.js:1875:2434)
    at Ke (main.RMJQDME5.js:1:1215)
    at tN.close (main.RMJQDME5.js:1954:7979)
    at tN.<anonymous> (main.RMJQDME5.js:1954:7930)
    at Generator.next (<anonymous>)
    at main.RMJQDME5.js:1:1395
    at new q (main.RMJQDME5.js:1875:2434)
```
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Plug een leaphy in een linux computer
2. Fix de permissions zodat er naar de serial port geschreven kan worden zonder elevated privileges:
    `sudo chmod 666 /dev/ttyUSB0` als je arduino nu op `/dev/ttyUSB0` zit
3. Open een chromium-based browser (firefox ondersteunt geen serial)
4. Ga naar [de website](https://leaphyeasybloqs.com/)
5. Open een workspace en de javascript console
6. Klik `Upload to robot`
7. Selecteer je serial port
8. Observeer de js console

**Expected behavior**
Het programma word naar de leaphy geupload

**Desktop (please complete the following information):**
 - OS: Linux (Arch linux voor ungoogled chromium en chromium, pop!_os (ubuntu-based) voor google chrome
 - Browser: ungoogled chromium, chromium, google chrome
 - Version: 122.0.6261.94-2, 122.0.6261.94, 121.0.6167.184 respectivelijk
 - Browser download locations: [ungoogled chromium komt van de AUR](https://aur.archlinux.org/packages/ungoogled-chromium-bin), [chromium uit de arch repo](https://archlinux.org/packages/extra/x86_64/chromium/), en [google chrome uit de pop!_os apt repo](https://apt.pop-os.org/) (geen directere link mogelijk)

**Additional context**
Als iemand meer informatie nodig heeft lever ik die graag.

rmoesbergen, 2024-03-03:
Hoi @Vosjedev . We hebben best veel gebruikers die linux gebruiken, bijvoorbeeld op chromebooks, en dat gaat eigenlijk prima. Ik gebruik het zelf ook. Een aantal tips om het goed werkend te krijgen:

1. Voeg jezelf toe aan de group waarin TTY devices zitten, meestal heet deze 'dialout' op ubuntu/debian based systemen (usermod -aG dialout <jouwusernaam>, daarna rebooten (ja echt))
2. Verwijder de package 'brltty' van je systeem, deze houdt alle ttyUSB/ACM devices open 

Met deze 2 aanpassingen zou het goed moeten werken.

Vosjedev, 2024-03-03:
Voor zover ik weet is het enige wat er gebeurt als ik mezelf aan de `dialout` group toevoeg is dat ik permissions krijg, dus dat is identiek aan `chmod +x /dev/ttyUSB0`. `brltty` is al verwijderd, anders zou `/dev/ttyUSB0` uberhaubt niet bestaan.

rmoesbergen, 2024-03-03:
> Voor zover ik weet is het enige wat er gebeurt als ik mezelf aan de `dialout` group toevoeg is dat ik permissions krijg, dus dat is identiek aan `chmod +x /dev/ttyUSB0`.

Met chmod +x maak je de device node executable, dat is niet hetzelfde als in de group dialout gezet worden. Het device moet writable zijn voor jouw user/group, wat je dus doet door jezelf in die group te zetten. Alleen uitloggen is niet voldoende, je moet echt rebooten.

Als dat allemaal niet werkt, is het een beetje een raadsel. Ik denk wel dat dit een lokaal issue is, en niet iets in de leaphy webbased code, aangezien het voor anderen prima werkt.

Vosjedev, 2024-03-03:
Pardon, ik bedoel `chmod 666 /dev/ttyUSB0`. Ik had net dingen met executables zitten doen, dus ik zat even verkeerd te denken. Op het (pop!_os) systeem (dat staat ook in de repo info van de orginele report) had ik chmod 666 gedaan, of een udev rule die hetzelfde doet in het geval van de arch linux.

rmoesbergen, 2024-03-05:
Werkt het wel met de Arduino IDE?

Vosjedev, 2024-03-05:
Ja, dat werkt. Ook met de arduino extension voor vscode werkt het, en minicom kan ook met hem communiceren.