leaphy-robotics/leaphy-webbased #250: [enhancement] give some love to the UI

koen1711, 2024-04-25:
Like here
[image]
Or here
[image]
Or variables
[image]
[image]
[image]

rmoesbergen, 2024-04-25:
or:
[image]

koen1711, 2024-05-01:
On hold till completion of #252 
@sverben